AutonomyLab/bebop_autonomy #146: Bebop2 doesn't give me the GPS after taking off.

akhkh, 2017-11-20:
Hi.

I made my own navigation code that subscribe "/bebop/fix" topic.

It always works when bebop2 is not flying. But after bebop2 taking off, the data (latitude, longitude, altitude) is fixed. ("/bebop/fix" topic)
I can subscribe the fixed GPS data, but it doesn't change.

I thought that ardrone3_pilotingstate_positionchanged_ptr does not change when bebop2 is flying.

The part is that bebop_driver/src/bebop_driver_nodelet, line 563. "gps_state_ptr = bebop_ptr_->ardrone3_pilotingstate_positionchanged_ptr->GetDataCstPtr(); "

How can I get the recent GPS data when bebop2 is flying?

Voidminded, 2018-02-07:
The `/bebop/fix` topic reports the home position of the drone used for return home functionality, of course it will record the latest TakeOff position as its home position.

If you want to read the GPS data you should subscribe to `/bebop/states/ardrone3/PilotingState/PositionChanged` topic which reports the current GPS data (while flying and landed) at 5Hz.

Note: If the number of satellites detected by GPS module is less than 3 the position data is invalid and equal to:
```
latitude: 500.0
longitude: 500.0
altitude: 500.0
```
You can check the number of detected satellites from `/bebop/states/ardrone3/GPSState/NumberOfSatelliteChanged` topic.

fabrizioschiano, 2019-08-28:
@Voidminded Could you confirm that you are able to access the GPS data at 5Hz on the Bebop?