AutonomyLab/bebop_autonomy #152: Proble while running the driver

dronemaster13, 2018-01-19:
When I am running the Driver the below errors are getting.
OS: Ubuntu 16.04
Drone: Bebop 2 with firmware 4.4.0 
Kindly help me to solve the issues, Thanks in advance!

started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [31757]
process[bebop/robot_state_publisher-3]: started with pid [31761]
[ INFO] [1516273858.812449808]: Initializing nodelet with 4 worker threads.
[ INFO] [1516273858.889269654]: [BebopSDK] 16:40:58:889 | Bebop:225 - Bebop Cnstr()
[ INFO] [1516273858.889355021]: Nodelet Cstr
[ INFO] [1516273858.896694000]: Connecting to Bebop ...
[ INFO] [1516273858.901361842]: [CB] 16:40:58:901 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1516273858.903311903]: [CB] 16:40:58:903 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1516273858.907468329]: [CB] 16:40:58:907 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1516273858.909197711]: [CB] 16:40:58:909 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1516273858.910856655]: [CB] 16:40:58:910 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1516273858.912449117]: [CB] 16:40:58:912 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1516273858.914268255]: [CB] 16:40:58:914 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1516273858.920218073]: [CB] 16:40:58:920 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1516273858.923245421]: [CB] 16:40:58:923 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1516273858.925856636]: [CB] 16:40:58:925 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1516273858.929676623]: [CB] 16:40:58:929 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1516273858.933694677]: [CB] 16:40:58:933 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1516273858.938797884]: [CB] 16:40:58:938 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1516273858.940921910]: [CB] 16:40:58:940 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1516273858.942376885]: [CB] 16:40:58:942 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1516273858.946924124]: [CB] 16:40:58:946 | CommonFlightPlanStateAvailabilityStateChanged:1457 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1516273858.949615314]: [CB] 16:40:58:949 | CommonFlightPlanStateComponentStateListChanged:1508 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1516273858.960734755]: [ARCONTROLLER_Network] 16:40:58:960 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1516273858.960914504]: [ARCONTROLLER_Stream2] 16:40:58:960 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1516273858.960950335]: [ARCONTROLLER_Stream2] 16:40:58:960 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ INFO] [1516273858.996339105]: [BebopSDK] 16:40:58:996 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 31910
[ERROR] [1516273859.031568031]: [ARCONTROLLER_Network] 16:40:59:031 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PilotingSettingsState.UNKNOWN -> Unknown 
[ERROR] [1516273859.128744750]: [ARCONTROLLER_Network] 16:40:59:128 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 UNKNOWN -> Unknown command
[ERROR] [1516273859.211262047]: [ARCONTROLLER_Network] 16:40:59:211 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PilotingState.UNKNOWN -> Unknown command
[ERROR] [1516273859.233692967]: [ARCONTROLLER_Network] 16:40:59:233 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PilotingState.UNKNOWN -> Unknown command
[ERROR] [1516273859.235011856]: [ARCONTROLLER_Network] 16:40:59:234 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 3 ARDrone3.PilotingState.UNKNOWN -> Unknown command
[ERROR] [1516273859.236224032]: [ARCONTROLLER_Network] 16:40:59:236 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 UNKNOWN -> Unknown command
[ERROR] [1516273859.243919060]: [ARCONTROLLER_Network] 16:40:59:243 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.UNKNOWN -> Unknown command
[ INFO] [1516273859.246298253]: [BebopSDK] 16:40:59:246 | Connect:321 - BebopSDK inited, lwp_id: 31757
[ WARN] [1516273859.246440592]: Resetting all settings ...
[ INFO] [1516273859.246601882]: [BebopSDK] 16:40:59:246 | ResetAllSettings:409 - All settings of the drone have been reset to default values.
[ERROR] [1516273859.253004429]: [ARCONTROLLER_Network] 16:40:59:252 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 UNKNOWN -> Unknown command
[ INFO] [1516273859.620783720]: Value for PilotingSettingsMaxAltitudeCurrent recved: 30
[ INFO] [1516273859.620922312]: [CB] 16:40:59:620 | Update:111 - Checking if PilotingSettingsMaxAltitudeCurrent exists in params ...
[ INFO] [1516273859.623297191]: [CB] 16:40:59:623 | Update:114 -   No
[ INFO] [1516273859.634580935]: Value for PilotingSettingsMaxDistanceValue recved: 2000
[ INFO] [1516273859.634629518]: [CB] 16:40:59:634 | Update:351 - Checking if PilotingSettingsMaxDistanceValue exists in params ...
[ INFO] [1516273859.635439525]: [CB] 16:40:59:635 | Update:354 -   No
[ INFO] [1516273859.655250156]: Value for PilotingSettingsNoFlyOverMaxDistanceShouldnotflyover recved: 0
[ INFO] [1516273859.655315558]: [CB] 16:40:59:655 | Update:431 - Checking if PilotingSettingsNoFlyOverMaxDistanceShouldnotflyover exists in params ...
[ INFO] [1516273859.656405167]: [CB] 16:40:59:656 | Update:434 -   No
[ERROR] [1516273859.670301326]: [ARCONTROLLER_Network] 16:40:59:670 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PilotingSettingsState.UNKNOWN -> Unknown 
[ INFO] [1516273859.681190342]: Value for PilotingSettingsMaxTiltCurrent recved: 20
[ INFO] [1516273859.681302639]: [CB] 16:40:59:681 | Update:191 - Checking if PilotingSettingsMaxTiltCurrent exists in params ...
[ INFO] [1516273859.684415973]: [CB] 16:40:59:684 | Update:194 -   No
[ INFO] [1516273859.699505154]: Value for PilotingSettingsAbsolutControlOn recved: 0
[ INFO] [1516273859.699559346]: [CB] 16:40:59:699 | Update:271 - Checking if PilotingSettingsAbsolutControlOn exists in params ...
[ INFO] [1516273859.700434825]: [CB] 16:40:59:700 | Update:274 -   No
[ INFO] [1516273859.717490953]: Value for PilotingSettingsBankedTurnValue recved: 0
[ INFO] [1516273859.717554442]: [CB] 16:40:59:717 | Update:511 - Checking if PilotingSettingsBankedTurnValue exists in params ...
[ INFO] [1516273859.718608385]: [CB] 16:40:59:718 | Update:514 -   No
[ INFO] [1516273859.730498527]: Value for SpeedSettingsMaxVerticalSpeedCurrent recved: 1
[ INFO] [1516273859.730560011]: [CB] 16:40:59:730 | Update:991 - Checking if SpeedSettingsMaxVerticalSpeedCurrent exists in params ...
[ INFO] [1516273859.731624403]: [CB] 16:40:59:731 | Update:994 -   No
[ INFO] [1516273859.746777225]: Value for SpeedSettingsMaxRotationSpeedCurrent recved: 100
[ INFO] [1516273859.747001298]: [CB] 16:40:59:746 | Update:1071 - Checking if SpeedSettingsMaxRotationSpeedCurrent exists in params ...
[ INFO] [1516273859.749236981]: [CB] 16:40:59:749 | Update:1074 -   No
[ INFO] [1516273859.761935256]: Value for SpeedSettingsHullProtectionPresent recved: 0
[... 189 lines cut ...]
[... cut ...]
[ERROR] [1516273859.235011856]: [ARCONTROLLER_Network] 16:40:59:234 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 3 ARDrone3.PilotingState.UNKNOWN -> Unknown command
[ERROR] [1516273859.236224032]: [ARCONTROLLER_Network] 16:40:59:236 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 UNKNOWN -> Unknown command
[ERROR] [1516273859.243919060]: [ARCONTROLLER_Network] 16:40:59:243 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.UNKNOWN -> Unknown command

dronemaster13, 2018-02-03:
@thomas-bamford Could you please help me for solving this bug? Thanks

Voidminded, 2018-02-09:
Hi @dronemaster13 

I finally was able to reproduced this error by launching the driver while my phone was also connected to the drone. I tried several times and noticed that the same issue happen _randomly_ when more than one WiFi device is connected to the drone (specially if one of them is my phone even if the FreeFlight app is closed).  
Make sure your phone or any WiFi device nearby other than the computer you are running the driver on does not automatically connect to the Bebop's WiFi.

dronemaster13, 2018-02-10:
Thanks @Voidminded . Thank you very much for the reply. I tried this also, but the same error happens. This is only associated with the Bebop2 drone. If the drone is bebop 1 it works well.
Currently I am not actively using the bebop1, so could you please help me for solving this for bebop2?