AutonomyLab/bebop_autonomy #155: image_raw window freeze when control form joystick

wz-eric, 2018-02-18:
In one terminal: roslaunch bebop_tools bebop_nodelet_iv.launch
In another :roslaunch bebop_tools joystick.launch

When starting control from the xbox360 joystick, the image_raw window freezes.

Anyone meet this problem?

wz-eric, 2018-02-18:
On ubuntu 16.04

ancientbees, 2018-03-14:
I am having the same problem, but on Ubuntu 10.04.

thomakl, 2018-05-11:
Hi,
I'm on Ubuntu 16.04.
I'm having the same issue after controlling from joystick (but not right after it, can't find the trigger of it).
Thank you

thomakl, 2018-05-11:
Hi,

Does anyone found a way to video stream directly (VLC, OpenCV or anything else) from the Bebop drone in order to use OpenCV ?

Thank you for your time,

thomakl, 2018-05-15:
Hi, 

I found this on another topic issue and it worked.

> Hello,
> Please try the following, in one terminal
> roslaunch bebop_driver bebop_nodelet.launch
> and then in a new terminal
> rosrun image_view image_view image:=/bebop/image_raw

Hope it could help you too,
Have a nice day,

TL