AutonomyLab/bebop_autonomy #157: How Can I update the Bebop Autonomy to the latest arsdk 3.13.1?

dronemaster13, 2018-03-05:
Can anybody  help me to update the current bebop_autonomy driver to the latest arsdk compatible?
I followed the procedures given in the [documentation](http://bebop-autonomy.readthedocs.io/en/latest/dev.html)  , but it gives many errors. It does not tells about the develop environment and procedures.

thomas-bamford, 2018-03-05:
Hi @dronemaster13, I can help with this upgrade. Could you post the errors?

dronemaster13, 2018-03-08:
@thomas-bamford Thanks for the reply and sorry for the delay. 
First I tried to upgrade [parrot_arsdk](https://github.com/AutonomyLab/parrot_arsdk) wrapper, But got the below error.

~/Desktop/parrot_arsdk-indigo-devel$ ./script/download_and_strip_arsdk.sh 
Temporary workspace: /tmp/tmp.ktt05oY9xL
Output: /home/gokul/Desktop/parrot_arsdk-indigo-devel/arsdk_3_13_1_p0_stripped.tgz
Get https://gerrit.googlesource.com/git-repo/clone.bundle
Get https://gerrit.googlesource.com/git-repo
remote: Finding sources: 100% (11/11)
remote: Total 11 (delta 0), reused 11 (delta 0)
Unpacking objects: 100% (11/11), done.
From https://gerrit.googlesource.com/git-repo
   685320b..1f36570  master     -> origin/master
Get https://github.com/Parrot-Developers/arsdk_manifests.git
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
remote: Counting objects: 125, done.        
remote: Total 125 (delta 0), reused 0 (delta 0), pack-reused 125        
Receiving objects: 100% (125/125), 26.30 KiB | 0 bytes/s, done.
Resolving deltas: 100% (51/51), done.
From https://github.com/Parrot-Developers/arsdk_manifests
 * [new branch]      master     -> origin/master
 * [new tag]         ARSDK3_version_3_10_0 -> ARSDK3_version_3_10_0
 * [new tag]         ARSDK3_version_3_10_1 -> ARSDK3_version_3_10_1
 * [new tag]         ARSDK3_version_3_11_0 -> ARSDK3_version_3_11_0
 * [new tag]         ARSDK3_version_3_12_0 -> ARSDK3_version_3_12_0
 * [new tag]         ARSDK3_version_3_12_1 -> ARSDK3_version_3_12_1
 * [new tag]         ARSDK3_version_3_12_3 -> ARSDK3_version_3_12_3
 * [new tag]         ARSDK3_version_3_12_4 -> ARSDK3_version_3_12_4
 * [new tag]         ARSDK3_version_3_12_5 -> ARSDK3_version_3_12_5
 * [new tag]         ARSDK3_version_3_12_6 -> ARSDK3_version_3_12_6
 * [new tag]         ARSDK3_version_3_13_0 -> ARSDK3_version_3_13_0
 * [new tag]         ARSDK3_version_3_13_1 -> ARSDK3_version_3_13_1
 * [new tag]         ARSDK3_version_3_8 -> ARSDK3_version_3_8
 * [new tag]         ARSDK3_version_3_8_1 -> ARSDK3_version_3_8_1
 * [new tag]         ARSDK3_version_3_8_2 -> ARSDK3_version_3_8_2
 * [new tag]         ARSDK3_version_3_8_3 -> ARSDK3_version_3_8_3
 * [new tag]         ARSDK3_version_3_9_0 -> ARSDK3_version_3_9_0
 * [new tag]         ARSDK3_version_3_9_1 -> ARSDK3_version_3_9_1
 * [new tag]         ARSDK3_version_3_9_2 -> ARSDK3_version_3_9_2

repo has been initialized in /tmp/tmp.ktt05oY9xL
Fetching project ulog
Fetching project apps_tools
Fetching project libARController
Fetching project libARMavlink
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total    Sp % Tent    Loeft  Speed
tal      % Re0ceiv    ed %  Xfe0rd   Aver ag e Sp eed 0  Ti    m 0  e     Tim e   0      0      T 0  i me  C   0u --:--rren:-- --:--:-- --:--:--     0t
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-  % Tota- l- - : - -%: -R-e c e i v e0d % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
    00          00        00          00        00          00            00            00  ----::----::----    00::0000::0022  ----::----::---        000 0                 0 00     0       0    00      0         0        000      0         0 00             0    0       00           0  0      0           0      0 0     ----0 ::  ----:0---:: --:------:   --  0- 00:::0:0-000-::00:3  003 -3  0-:--:0---:0-:--::0---3::--  --- - -    : -      0 -0
0:
-
-     0
curl: (22) Tcche rueqruel:s (t22ed) U RLThue  rerqurle: st(ed UetRL urre2tunerne2)d ed rro Tr: er404 Nheot  Froor:re quuned
ste404d N URL retuornt ed eFrroro: 404 Not Found
und
Server does not provide clone.bundle; ignoring.Server does not provide clone.bundle; ignoring.
Server does not provide clone.bundle; ignoring.

curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
remote: Counting objects: 3259, done.        
[... 369 lines cut ...]
Exporting ./packages.git/libARMedia ...
Exporting ./packages.git/libARNetwork ...
Exporting ./packages.git/libARCommands ...
Exporting ./packages.git/libARDataTransfer ...
Exporting ./packages.git/libARMavlink ...
Exporting ./packages.git/libARUtils ...
Exporting ./packages.git/libARStream2 ...
Exporting ./packages.git/arsdk-xml ...
Exporting ./packages.git/libtar ...
Exporting ./packages.git/libsdp ...
Exporting ./packages.git/libARDiscovery ...
Exporting ./packages.git/libARUpdater ...
Exporting ./packages.git/libfutils ...
Exporting ./packages.git/libpomp ...
Exporting ./packages.git/libARController ...
Exporting ./packages.git/libARSAL ...
Exporting ./packages.git/librtsp ...
Exporting ./packages.git/ARSDK3 ...
Exporting ./packages.git/mavlink ...
Exporting ./packages.git/Samples ...
Exporting ./packages.git/libARStream ...
Exporting ./packages.git/libARNetworkAL ...
Applying package patches ...
error: patch failed: packages/ARSDKTools/json-c/atom.mk:23
error: packages/ARSDKTools/json-c/atom.mk: patch does not apply

dronemaster13, 2018-03-14:
@thomas-bamford  Could you please help me to resolve this issue?

thomas-bamford, 2018-03-14:
Hey @dronemaster13, this appears to be caused by the urls used to pull the various sdk packages together. I will take a deeper look into this tonight.

dronemaster13, 2018-03-15:
Hi @thomas-bamford  Thanks for the reply. I am eagerly waiting for your next update.

dronemaster13, 2018-03-19:
Hi @thomas-bamford , Is there any update?

thomas-bamford, 2018-03-19:
Ok, so the issue at this step was that the json-c_avoid_so_version.patch file in parrot_arsdk/patch was not up to date with the latest ARSDK. If you update it to the attached file below (and remove the file extention .txt) then the download_and_strip_arsdk.sh step will finish without errors.

[json-c_avoid_so_version.patch.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1826410/json-c_avoid_so_version.patch.txt)

However, now when I move on to the next step of building the parrot_arsdk package in catkin, there are more errors arising. See attached file here:

[build_errors.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1826428/build_errors.txt)

This seems to be due to changes from using avahi without dbus to using avahi with dbus as a dependency in the libARDiscovery module. Will spend more time on it today.

thomas-bamford, 2018-03-19:
If you want to help fixing the issue, I have forked parrot_arsdk to my account to keep track of my changes. It does not build yet, but all the steps up to `catkin build` are finished.

https://github.com/thomas-bamford/parrot_arsdk

dronemaster13, 2018-03-20:
@thomas-bamford Great!. Thanks again. 
I am waiting for your next update.

dronemaster13, 2018-04-05:
@thomas-bamford Is there any updates? Thank you.

thomas-bamford, 2018-04-06:
I have been preparing and conducting field experiments over the past few weeks and have not had time to advance. Again, if you want to help fix the issue, please see:

https://github.com/thomas-bamford/parrot_arsdk

dronemaster13, 2018-04-09:
@thomas-bamford I tried my level best to solve this issue, But I was not able to make it.
Can you please give any advises or  directions, where I have to look into?

dronemaster13, 2018-06-07:
@thomas-bamford I hope that you are comparatively free now. Could you please look again to the issue?

dronemaster13, 2018-07-02:
Hi, Anybody please help for upgrading the Bebop autonomy to latest ARSDK....Thanks  in advance.

thomas-bamford, 2018-07-03:
Almost done the upgrade. Pull request coming soon.

thomas-bamford, 2018-07-03:
See https://github.com/AutonomyLab/bebop_autonomy/pull/168 and https://github.com/AutonomyLab/parrot_arsdk/pull/9. These compile together on Ubuntu 16.04 and will be tested on the vehicles in a few hours.