AutonomyLab/bebop_autonomy #160: Parrot Bebop 2 System ID

air4ce, 2018-04-30:
I'm trying to model the Parrot Bebop 2 and I was wondering if a mathematical model already exists.  

If not, I want to model the Bebop 2 through solving a parameter optimization.  To do this, I need some way to record the inputs to the Bebop itself during a commanded maneuver (e.g. rotor RPM or commanded motor voltage).  My question is whether not there's an existing way of measuring and recording this information for a standard Bebop 2?  Finally, this information doesn't need to be transmitted in real time and can be collected after the fact.

Voidminded, 2018-05-09:
Hi @air4ce 

AFAIK Parrot refused to share their mathematical model with us.
@mani-monaj approximated a first-order non-linear model for converting Forward/Lateral linear velocities to Pitch/Roll of the drone using another method which is used in [AutonomyLab/bebop_vel_ctrl: Velocity controller for Parrot Bebop Drone (ROS package)](https://github.com/AutonomyLab/bebop_vel_ctrl).  For more information about this dynamic model and how we have calculated it, you can refer to our [paper](http://autonomy.cs.sfu.ca/doc/monajjemi_iros16.pdf) or more detailed in [Mani's Ph.D. thesis](http://autonomy.cs.sfu.ca/doc/monajjemi_phd.pdf).

If you prefer to record the motor's RPM to calculate the dynamic model, you have several options:
- Optical RPM sensors: I personally am not a fan of these sensors, you need to add one for each motor, mark one of the prop's blades with reflector tapes and literally count the RPM, basically equip each prop with a tachometer.
- Eagle Tree BLDC RPM sensor: If you don't mind tapping into the drone's hardware, you can connect one of three motor inputs to these sensors to count the RMP. They are very light weighted and easy to use, you can drive all 4 using a single Arduino on board the drone. I usually equip my custom designed drones with these sensors. However, you need to know the number of poles of the motor to convert the output frequency to RPM.
- FrSky smart port RPM sensor: These sensors need to be tapped into ESC's output as well (usually work with tapping into one of ESC's outputs but if you use both sensor's inputs on any two different phases of motor you will get a more reliable and better result). These sensors provide VERY accurate results, but they only communicate over FrSky SmartPort, so you will either need to use a FrSky RC receiver module (I used X8R) onboard to connect the sensors to which translate the information received on SmartPort, or crack their protocol to read the info on SmartPort using another onboard computer.

Anyway you need to equip the drone with 4 sensors and a separate processor for this task.

davidvar, 2018-05-17:
Hello. Voidminded

 this [link](https://bebop-autonomy.readthedocs.io/en/latest/piloting.html) explains how to calculate the roll and pitch angles based on the linear velocity x and y respectively. The equations are:

roll_degree       = linear.y  * max_tilt_angle
pitch_degree      = linear.x  * max_tilt_angle

but those equations didn't match with your paper. can you explain please how the dynamic model works?

Voidminded, 2018-05-24:
Hi @davidvar 

I am sorry for the confusion we may have caused in the documentation. The [geometry_msgs/Twist message in ROS](http://docs.ros.org/api/geometry_msgs/html/msg/Twist.html) has two components, linear and angular. The equations you mentioned from our documents are only for mapping the Twist message to Roll and Pich. For instance `linear.x` multiplied by the constant coefficient `max_tilt_angle` provides the Pitch value for the drone, but it does not mean the drone will move at the linear rate you have set for `linear.x` in the forward direction. In the aforementioned paper, we explained how to calculate the Roll/Pitch based on desired Lateral/Forward velocities considering the dynamic model of the UAV (the input for the Twist message).

As for why Roll and Pith are sent over the linear component of Twist message, not the angular one, I believe it's because the Twist message is supposed to provide the speed values, Yaw value sent on `angular.z` provide angular rotation velocity, but the Pitch and Roll values are the angular position of the body frame which transform to the linear velocity of body frame thus are published on linear component of Twist message. (`angular.x` and `angular.y` are reserved for Acro mode).

davidvar, 2018-06-01:
Hi. @Voidminded

Thank you for the reply. What I understand based on your response and the information provided is as follows: for example, if I propose a position and velocity controller, the output of the controller must be multiplied by the dynamic model?
In detail I attach an image [https://drive.google.com/open?id=1lb0xzGAKg3i5Po_-B_1t7pJkRNGLvpgw](url) is my scheme correct?