AutonomyLab/bebop_autonomy #162: Can't move the drone with controller

thomakl, 2018-05-09:
Hi,

The installation of Bebop_autonomy, ros, the build and launching "roslaunch"  worked well.

My problem is : I can connect with the drone, see through its lens, make him take-off, land (everything works with buttons) ; except that as soon I try to make it move with the thumb pad axis, I got some error of ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 
(I configured the controller thanks to joy_teleop config)

----------------------------
[STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1525872125.413915428]: [ARCONTROLLER_Network] 15:22:05:413 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1525872125.414044911]: [ARCONTROLLER_Stream2] 15:22:05:414 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1525872125.414063692]: [ARCONTROLLER_Stream2] 15:22:05:414 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ INFO] [1525872125.455834316]: [BebopSDK] 15:22:05:455 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 20692
[ERROR] [1525872125.487058271]: [ARCONTROLLER_Network] 15:22:05:487 | ARCONTROLLER_Network_ReaderRun:1288 - **ARCOMMANDS_Decoder_DecodeBuffer () failed : 2** ARDrone3.PilotingSettingsState.UNKNOWN -> Unknown 
[ERROR] [1525872125.581168052]: [ARCONTROLLER_Network] 15:22:05:581 | ARCONTROLLER_Network_ReaderRun:1288 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 UNKNOWN -> Unknown command
[ERROR] [1525872125.667081319]: [ARCONTROLLER_Network] 15:22:05:667 | ARCONTROLLER_Network_ReaderRun:1288 - **ARCOMMANDS_Decoder_DecodeBuffer () failed :** 3 ARDrone3.PilotingState.UNKNOWN -> Unknown command
[ERROR] [1525872125.673971860]: [ARCONTROLLER_Network] 15:22:05:673 | 
[ INFO] [1525872125.677443252]: [BebopSDK] 15:22:05:677 | Connect:321 - BebopSDK inited, lwp_id: 20531
[ WARN] [1525872125.677511466]: Resetting all settings ...
[ INFO] [1525872125.677578768]: [BebopSDK] 15:22:05:677 | ResetAllSettings:417 - All settings of the drone have been reset to default values.
---------------------------------------

I try to install parrot_ardsk in the meanwhile : still the same problem.
Can't figure it out why.

Thank you for your time

thomakl, 2018-05-10:
Hi @Voidminded,

Thank you for your quick answer, I appreciate.
Here are the config files : I used the original log710 config to adapt to my PS3 like controller (here under the name "PS3-LIke", but in the config folder I let the original name "log710").
I don't know if it can help you to look at the hoy_teleop.launch but I didn't touch it.
Also I need to mention that when I launch the jstest and joy test to see if all functions of the controller worked : it does. Axis and buttons respond correctly.

[PS3-Like.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1990689/PS3-Like.txt)
[log710.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1990692/log710.txt)
[joy_teleop.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1990740/joy_teleop.txt)

Thank you for your time,
Have a nice day.

TL

Voidminded, 2018-05-10:
@thomakl I don't have PS3 joystick, but tried your config with a PS4 joystick, the axes were not correctly mapped. Remember in PS joysticks axis 3 and button 6 are being controlled by the same switch and axis 4 and button 7 likewise.

Here I fixed you config for you which works now, I've created a `ps4.yaml` config file mapping the same config on Logitech to PS4 joystick which I'll add to the repo.

[PS3-Like.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1993517/PS3-Like.txt)
[ps4.txt](https://github.com/AutonomyLab/bebop_autonomy/files/1993518/ps4.txt)

thomakl, 2018-05-11:
Hi @Voidminded,

I din't understand that I need to press the R2 button to active the axis stick.
Then I only needed to change the mapping for the right thumb stick.
It worked perfectly now.
One problem left to fix is the streaming video that freezed after a moment : That will be for another post issue.

Thank you again for your help.

TL