AutonomyLab/bebop_autonomy #166: Install bebop autonomy on ROS bionic ubuntu 18.04

h3ct0r, 2018-06-12:
Hi!

Any plans to make this repo compatible with **ROS bionic** running on Ubuntu 18.04?

Thanks!

thomas-bamford, 2018-06-18:
Yes, we are really hoping to upgrade this package to work under Ubuntu 18.04.

thomas-bamford, 2018-07-03:
See https://github.com/AutonomyLab/bebop_autonomy/pull/168 and https://github.com/AutonomyLab/parrot_arsdk/pull/9. These compile together on Ubuntu 16.04 and will be tested on the vehicles in a few hours. Will try making this compatible with ROS Melodic on Ubuntu 18.04 in my spare time this week.

thomas-bamford, 2018-07-05:
PR #168 compiles with ROS Melodic using changes introduced in PR #164. Note: joy and joy_teleop (dependencies of bebop_tools package) are not yet released in ROS Melodic so they have to be built from their source repositories.

thomas-bamford, 2018-08-09:
Waiting for parrot_arsdk binary to be released by ROS so that PR #168 can be merged.

bebe04, 2018-09-25:
I have the same problem, I have a mistake : "bebop_tools: Cannot locate rosdep definition for [joy_teleop]" 
Can you tell me how get out of this problem ?
For tout information I want to install melodic ROS on Ubuntu 18.04 and then install bebop_autonomie

thomas-bamford, 2018-09-26:
The joy_teleop binary package is not yet available for ROS Melodic but you can install it from source.
`cd ~/bebop_ws/src`
`git clone https://github.com/ros-teleop/teleop_tools.git`
`cd ~/bebop_ws`
`catkin build`

Kluchozaur, 2018-11-08:
Hi! Is it possible to run it on Ubuntu 18.04 with ROS Melodic?

Ceepert, 2019-11-25:
This hasnt been aktive for a long time, is bebop autonomy not installable on melodic?

gsilano, 2019-11-25:
> Hi! Is it possible now to run it on Ubuntu 18.04 with ROS Melodic? I need gazebo 8 version, but I need this driver too ;)

Gazebo 8/9 is available also for the Kinetic Kame release of ROS. If you need to test your control algorithm with bebop_autonomy and Gazebo, you can think to use ROS Kinetic on Ubuntu 16.04. It is also possible to install the ROS distro on Ubuntu 18.04, but you have to manage manually all the steps.

Ceepert, 2019-11-25:
> 
> 
> > Hi! Is it possible now to run it on Ubuntu 18.04 with ROS Melodic? I need gazebo 8 version, but I need this driver too ;)
> 
> Gazebo 8/9 is available also for the Kinetic Kame release of ROS. If you need to test your control algorithm with bebop_autonomy and Gazebo, you can think to use ROS Kinetic on Ubuntu 16.04. It is also possible to install the ROS distro on Ubuntu 18.04, but you have to manage manually all the steps. In the following a guide:
> 
> https://medium.com/@abhiksingla10/setting-up-ros-kinetic-and-gazebo-8-or-9-70f2231af21a

Does that also work to install kinetic on Ubuntu 18.04?

gsilano, 2019-11-25:
It should work! Try, and let us know!!

anthonymiglio, 2020-12-14:
**What will be the fate of bebop autonomy, now that ROS Kinetic is no longer supported?**

Have anyone manage to install bebop autonomy on ROS Melodic on Ubuntu 18.04?

AyoubTHABIT, 2021-01-30:
> 
> 
> **What will be the fate of bebop autonomy, now that ROS Kinetic is no longer supported?**
> 
> Have anyone manage to install bebop autonomy on ROS Melodic on Ubuntu 18.04?

PLease, did you find an answer for that? or some other package for bebop suitable for ubuntu 18.04?

widmanhg, 2024-08-22:
did anyone achieve anything?