AutonomyLab/bebop_autonomy #167: Sphinx and Bebop_autonomy

amstalfo, 2018-07-02:
I have seen a video on youtube that shows a person using bebop_autonomy and Sphinx together. http://forum.developer.parrot.com/t/using-bebop-autonomy-with-sphinx-on-same-machine/6726 highlights the problem. When I run the bebop_node.launch or bebop_nodelet.launch file I get the same error as I would if the real quadcopter were off. I am using the bebop2 and ip address of 10.202.0.1 for sphinx instead of the default ip address for the bebop. Can someone lead me in the right direction to using bebop_autonomy and Sphinx together? Thanks

thomas-bamford, 2018-07-31:
In Sphinx's documentation (https://developer.parrot.com/docs/sphinx/dronefile.html) it states that in the dronefile:

"For the currently supported drones,

    NAME_ONCE_ASSIGNED must be ‘eth0’,
    IP_ADDRESS must be ‘192.168.42.1/24’."

So you should change your configurations in the ros launch file and simulator dronefile, accordingly. This is regardless of the IP address found when running 'iwconfig.' Unlike the Bebop 2 hardware, it seems that the only IP address configuration allowed is the default '192.168.42.1'.

suraj2596, 2019-05-11:
Another possible reason would be the GPU support. The built-in GPU(Intel-xxx) doesn't support most of the time even with `low_gpu="false"`. I had to migrate to another laptop with 1060 to get it working.

gsilano, 2019-05-11:
As discussed on the [Parrot Bebop forum](https://forum.developer.parrot.com/t/how-to-get-drone-status-and-parameters/8697/5), me and my colleagues are working on Sphinx and bebop_autonomy, as well. We made available [our code on GitHub](https://github.com/gsilano/BebopS), the dev/Sphinx branch, but it is not yet ready to be released. In any case, I thought it would be useful for someone, so please do not hesitate to contact me if you have any question.