AutonomyLab/bebop_autonomy #170: roslaunch bebop_nodelet.launch fails on 18.04/Melodic/src

anqixu, 2018-07-12:
I get the following error when roslaunching both the node and the nodelet:

> [ERROR] [1531415229.229082225]: Failed to load nodelet [/bebop/bebop_nodelet] of type [bebop_driver/BebopDriverNodelet] even after refreshing the cache: Failed to load library /home/mimic/catkin_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)

The culprit is `libcommands.so` was not found in `$LD_LIBRARY_PATH`, which only contains `<CATKIN_WS>/devel/lib` and not `<CATKIN_WS>/devel/lib/parrot_arsdk`.

I'm rusty with my catkin skills, and trying to change [the install dir](https://github.com/AutonomyLab/parrot_arsdk/blob/indigo-devel/CMakeLists.txt#L71) didn't work (after a `catkin clean` and `catkin build`).

I can bypass this issue in two ways:
- add `<CATKIN_WS>/devel/lib/parrot_arsdk` to `$LD_LIBRARY_PATH` (e.g. in `~/.bashrc`),
- or duplicating the symbolic links via `cd <CATKIN_WS>/devel/lib/parrot_arsdk && cp -av lib* ..`

If you guys can fix this issue properly without using the bypass hacks above, then great! Otherwise, I hope that others might find these hacks helpful.

captjulian, 2018-07-24:
Same issue here

WinKILLER, 2018-09-22:
Same problem. Any progress?

TOTON95, 2018-10-09:
Had to copy the entire library into the devel/lib/ of the driver to make it work, if anyone is working with other type of Parrot Drones or products (rossumo, for example), this is also a dirty workaround that can work.

WiPhi7600, 2019-03-16:
Any one crack this yet? Having the same issue. @TOTON95 What do you mean by your response if you don't mind elaborating? I'm very new to ROS.

harrafa, 2019-08-27:
> @harrafa <https://github.com/harrafa>, it has to point to the
> parrot_arsdk in ROS General library, I don't have my computer at hand. But
> I ended up copy and pasted the libraries into bebop/devel/lib.

Thanks a lot Alexis. This makes so much more sense.
Will update you when I am able to get through:)
Warm regards,
Taizoon.

leithw2, 2020-02-20:
> I get the following error when roslaunching both the node and the nodelet:
>
>
> The culprit is `libcommands.so` was not found in `$LD_LIBRARY_PATH`, which only contains `<CATKIN_WS>/devel/lib` and not [...]

Problem solved thanks

gsilano, 2020-02-20:
Could you explain how you solved the issue? Thanks!

leithw2, 2020-02-20:
I only use the "bypass hacks" mentioned by @anqixu 

> add <CATKIN_WS>/devel/lib/parrot_arsdk to $LD_LIBRARY_PATH (e.g. in ~/.bashrc),

Ceepert, 2020-02-27:
> I only use the "bypass hacks" mentioned by @anqixu
> 
> > add <CATKIN_WS>/devel/lib/parrot_arsdk to $LD_LIBRARY_PATH (e.g. in ~/.bashrc),

could you go i little bit into more detail?
What is the exact command (Im new to using ubuntu)
I dont have a parrot_arsdk folder in catkin_workspace/devel/lib/   I have the ParrotSDK folder on my desktop? How do I get it into that folder?

anqixu, 2020-03-03:
* let's say your catkin workspace is located at `~/catkin_ws`
* for simplicity, let's export an environment variable:
    `$ export CATKIN_WS="~/catkin_ws"`
* to do this properly, you should open your `~/.bashrc` file in a text editor, and add a line somewhere (e.g. at the end) with the following content:
    `LD_LIBRARY_PATH=~/catkin_ws/devel/lib/parrot_arsdk:$LD_LIBRARY_PATH`
* but the quick and dirty way is:
    `$ echo LD_LIBRARY_PATH=$CATKIN_WS/devel/lib/parrot_arsdk:$LD_LIBRARY_PATH >> ~/.bashrc`

MatheusR42, 2020-04-08:
@Ceepert , I also could not find the parrot_arsdk in the folder. I manage to fix this way:

- Install the parrot_arsdk:
`sudo apt install ros-melodic-parrot-arsdk`

- Copy the folder from /opt/ros/melodic/lib/parrot_arsdk to <CATKIN_WS>/devel/lib/

- Update .bashrc (as @anqixu comment, remember to change to you workspace name) 
LD_LIBRARY_PATH=~/catkin_ws/devel/lib/parrot_arsdk:$LD_LIBRARY_PATH

leithw2, 2020-04-09:
After reinstalling ubuntu 18 I was facing this problem again, I used the one above and it still did not work, I realized I was using "catkin build" so I tried "Catkin make" and this solved the problem for me