AutonomyLab/bebop_autonomy #173: Anafi Support

skohlbr, 2018-09-04:
Searched for "Anafi" but did not find any existing issues. Are there any plans for adding support for the new Parrot Anafi? Before that can happen, it appears the SDK first has to be made available by Parrot, see [Parrot Anafi SDK Thread on the developer forum](https://forum.developer.parrot.com/t/parrot-anafi-sdk/7508/13).

espetro, 2019-05-09:
Hey there! The [Parrot SDK](https://developer.parrot.com/) has been officially updated this week. Now, existing modules for Bebop drones should be usable in Anafi.

fabrizioschiano, 2020-02-20:
Has anyone used this software with the Anafi?

espetro, 2020-02-20:
I didn't need it. The Parrot SDK does the work as of now. If you are willing to use ROS/2, you can interface with the SDK quite easily

fabrizioschiano, 2020-02-20:
@espetro , thanks for the quick reply. And what about ROS 1 ? Also, could you point me to the code you used for interfacing ROS2 with the Anafi  ?

MaxSmith13, 2020-04-01:
@fabrizioschiano Did you manage to get a solution to connect a Parrot Anafi to the bebop_autonomy SDK?

fabrizioschiano, 2020-04-01:
I was working with a colleague on this. He told me:
I ended up using Olympe inside my own ROS node, which was part of the reason to look into ROS+Python3.

However, Olympe is basically just a convenient wrapper around the (relatively new) arsdk-ng/arsdk-xml (see: https://github.com/Parrot-Developers/arsdk-ng), so any package based on that should also work. However, from what I could find out, the existing ROS packages for Parrot drones (e.g. bebop_autonomy) use a previous-gen SDK (ARSDK3) and won't work out-of-the-box with the Anafi.

MaxSmith13, 2020-04-01:
@fabrizioschiano Thank you for your explanation!

espetro, 2020-04-02:
Hi @fabrizioschiano @MaxSmith13 ,

Unfortunately I cannot share with you the code as it was done for a private company. Nevertheless, I can depict what I did.
As @fabrizioschiano has said, Olympe is a wrapper for the new version of the Parrot SDK, which is done in Python 3. Sadly, ROS uses Python 2.7 which needs a separate runtime, but [I've heard][uno] that they're going to do newest versions in Py3. My workaround was to create a **Olympe to ROS2 to ROS1 bidirectional channel** using `rosbridge`, so that I could still use most of the packages in ROS1 yet interfacing with the drone using Olympe.

Hope that this helps you in some way 😄

[uno]: https://discourse.ros.org/t/ros-1-and-python-2-7-end-of-life/9941