AutonomyLab/bebop_autonomy #174: bebp_aunonomy + opencv

shaarkrat2, 2018-10-01:
hello!

i'm trying to import the video stream from my drone bebop2 (using ros kinetic and bebop_autonomy) to opencv

 i'm connecting to the drone using the command 
      $ roslaunch bebop_tools bebop_nodelet_iv.launch

this command allows me to connect to the drone and to get the video stream in a new windows, but i can't import that stream in opencv.

I've installed opencv_bridge ( $ sudo apt-get install ros-kinetic-cv-bridge ) but I can't use it.
Does anyone know how to get the video stream in opencv?

Thanks

thomas-bamford, 2018-10-01:
Tutorials on using the cv_bridge package for coding in [C++](http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages) and [Python](http://wiki.ros.org/cv_bridge/Tutorials/ConvertingBetweenROSImagesAndOpenCVImagesPython). For one of my projects I use a modified version of the Python example.

shaarkrat2, 2018-10-01:
I doesn't understand how the cv bridge work
Do I have to create a new nodlet and suscribe to the video stream?