AutonomyLab/bebop_autonomy #181: Can't running the driver as a Node (Ubuntu 18.04)

Kluchozaur, 2018-11-09:
Hi everyone! I would like to use bebop_autonomy on Ubuntu 18.04. Installation works fine, but I have problem with running the driver as a node, here are my errors:

**milena@milena-GV62-8RC:~/bebop_ws$ roslaunch bebop_driver bebop_node.launch
... logging to /home/milena/.ros/log/3b91a410-e41e-11e8-83d2-1257a59fcfc4/roslaunch-milena-GV62-8RC-14943.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://milena-GV62-8RC:44069/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 10.202.0.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [14956]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3b91a410-e41e-11e8-83d2-1257a59fcfc4
process[rosout-1]: started with pid [14967]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [14974]
process[bebop/robot_state_publisher-3]: started with pid [14975]
[ INFO] [1541767914.203514213]: Initializing nodelet with 8 worker threads.
[ERROR] [1541767914.223188310]: Failed to load nodelet [/bebop/bebop_driver] of type [bebop_driver/BebopDriverNodelet] even after refreshing the cache: Failed to load library /home/milena/bebop_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
[ERROR] [1541767914.223247951]: The error before refreshing the cache was: Failed to load library /home/milena/bebop_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
[FATAL] [1541767914.223277259]: bebop_driver nodelet failed to load.
[bebop/bebop_driver-2] process has died [pid 14974, exit code 1, cmd /home/milena/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/milena/.ros/log/3b91a410-e41e-11e8-83d2-1257a59fcfc4/bebop-bebop_driver-2.log].
log file: /home/milena/.ros/log/3b91a410-e41e-11e8-83d2-1257a59fcfc4/bebop-bebop_driver-2*.log
^C[bebop/robot_state_publisher-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done**

How can i fix it?

closed 2018-11-09 by Kluchozaur

SchwesterKatze, 2019-03-06:
I have the same issue and my shared libraries like libbebop.so and libbebop_driver_nodelet.so are in the same folder <bebop_ws>/devel/lib. And is still having the same issue. Can you help me

WiPhi7600, 2019-03-17:
@Kluchozaur Did you ever find a solution?