AutonomyLab/bebop_autonomy #183: Catkin build fails

tahakhursheed, 2018-12-04:
catkin build fails, I am following steps from https://bebop-autonomy.readthedocs.io/en/latest/installation.html
I followed the same steps on an Ubuntu 16.04 VM. catkin build worked. Now, I am trying to execute catkin build on a native Ubuntu 16.04 with freshly installed ROS kinect. Any Suggestions?

`Errors     << bebop_driver:make /home/riotu/bebopta_ws/logs/bebop_driver/build.make.000.log             
CMakeFiles/bebop_driver_node.dir/src/bebop_driver_node.cpp.o: In function `main':
bebop_driver_node.cpp:(.text+0x6a): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
bebop_driver_node.cpp:(.text+0x9b): undefined reference to `ros::names::getRemappings()'
bebop_driver_node.cpp:(.text+0xc1): undefined reference to `ros::this_node::getName()'
bebop_driver_node.cpp:(.text+0x128): undefined reference to `nodelet::Loader::load(std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::vector<std::string, std::allocator<std::string> > const&)'
bebop_driver_node.cpp:(.text+0x15c): undefined reference to `nodelet::Loader::listLoadedNodelets()'
bebop_driver_node.cpp:(.text+0x24d): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
bebop_driver_node.cpp:(.text+0x364): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/riotu/bebopta_ws/devel/.private/bebop_driver/lib/bebop_driver/bebop_driver_node] Error 1
make[1]: *** [CMakeFiles/bebop_driver_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/riotu/bebopta_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
........................................................................................................
Failed     << bebop_driver:make                    [ Exited with code 2 ]                               
Failed    <<< bebop_driver                         [ 27.1 seconds ]                                     
Abandoned <<< bebop_tools                          [ Unrelated job failed ]                             
[build] Summary: 3 of 5 packages succeeded.                                                             
[build]   Ignored:   1 packages were skipped or are blacklisted.                                        
[build]   Warnings:  1 packages succeeded with warnings.                                                
[build]   Abandoned: 1 packages were abandoned.                                                         
[build]   Failed:    1 packages failed.                                                                 
[build] Runtime: 41.0 seconds total.                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`

chaosbuster, 2019-01-13:
I'm having the same problem.

chaosbuster, 2019-01-13:
I'm having the same problem in the create_autonomy repository for the iRobot iCreate base.

Balaji-Udayagiri, 2019-05-22:
I am facing similar issues following the mentioned site.
The Build summary is as follows:
[build] Summary: 3 of 5 packages succeeded.                           
[build]   Ignored:   1 packages were skipped or are blacklisted.      
[build]   Warnings:  1 packages succeeded with warnings.              
[build]   Abandoned: 1 packages were abandoned.                       
[build]   Failed:    1 packages failed.   

In file included from /home/balaji/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h:22:0,
                 from /home/balaji/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:43:
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:32:11: error: macro "max" requires 2 arguments, but only 1 given
     , max()
           ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:33:11: error: macro "min" requires 2 arguments, but only 1 given
     , min()
           ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:38:17: error: macro "max" requires 2 arguments, but only 1 given
     , max(_alloc)
                 ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:39:17: error: macro "min" requires 2 arguments, but only 1 given
     , min(_alloc)
                 ^
In file included from /opt/ros/kinetic/include/json-c/json.h:19:0,
                 from /opt/ros/kinetic/include/parrot_arsdk/libARDiscovery/ARDISCOVERY_Device.h:41,
                 from /opt/ros/kinetic/include/parrot_arsdk/libARDiscovery/ARDiscovery.h:38,
                 from /home/balaji/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:38:
/usr/include/c++/5/complex: In function ‘_Tp std::__complex_abs(const std::complex<_Tp>&)’:
/opt/ros/kinetic/include/json-c/bits.h:20:18: error: expected unqualified-id before ‘(’ token
 #define max(a,b) ((a) > (b) ? (a) : (b))
                  ^
/usr/include/c++/5/bits/locale_facets_nonio.tcc: In member function ‘_InIter std::__cxx11::time_get<_CharT, _InIter>::_M_extract_name(std::__cxx11::time_get<_CharT, _InIter>::iter_type, std::__cxx11::time_get<_CharT, _InIter>::iter_type, int&, const _CharT**, std::size_t, std::ios_base&, std::ios_base::iostate&) const’:
/opt/ros/kinetic/include/json-c/bits.h:16:18: error: expected unqualified-id before ‘(’ token
 #define min(a,b) ((a) < (b) ? (a) : (b))
                  ^
In file included from /home/balaji/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h:22:0,
                 from /home/balaji/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:43:
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h: In constructor ‘dynamic_reconfigure::ConfigDescription_<ContainerAllocator>::ConfigDescription_()’:
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:33:5: error: expected ‘(’ before ‘,’ token
     , min()
     ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:34:5: error: expected ‘(’ before ‘,’ token
     , dflt()  {
     ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h: In constructor ‘dynamic_reconfigure::ConfigDescription_<ContainerAllocator>::ConfigDescription_(const ContainerAllocator&)’:
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:39:5: error: expected ‘(’ before ‘,’ token
     , min(_alloc)
     ^
/opt/ros/kinetic/include/dynamic_reconfigure/ConfigDescription.h:40:5: error: expected ‘(’ before ‘,’ token
     , dflt(_alloc)  {
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
cd /home/balaji/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................
Failed     << bebop_driver:make                [ Exited with code 2 ] 
Failed    <<< bebop_driver                     [ 12.3 seconds ]       
Abandoned <<< bebop_tools                      [ Unrelated job failed ]
[build] Summary: 2 of 4 packages succeeded.                           
[build]   Ignored:   1 packages were skipped or are blacklisted.      
[build]   Warnings:  None.                                            
[build]   Abandoned: 1 packages were abandoned.                       
[build]   Failed:    1 packages failed.                               
[build] Runtime: 16.3 seconds total.