AutonomyLab/bebop_autonomy #188: Trouble with launching driver

WiPhi7600, 2019-03-16:
I know there is a similar post about this made a while ago, but considering there isn't a really detailed answer on how to solve this, I thought I might post this to try and get a more in depth solution.

I tried using the instructions here https://bebop-autonomy.readthedocs.io/en/latest/running.html and the preceding instructions. But I keep getting the following error:

will@will-VirtualBox:~/bebop_ws/devel$ roslaunch bebop_driver bebop_node.launch... logging to /home/will/.ros/log/a5e7ba98-47b0-11e9-a5e9-080027ad78cc/roslaunch-will-VirtualBox-17640.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://will-VirtualBox:45899/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /bebop/
	bebop_driver (bebop_driver/bebop_driver_node)
	robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [17653]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a5e7ba98-47b0-11e9-a5e9-080027ad78cc
process[rosout-1]: started with pid [17664]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [17671]
process[bebop/robot_state_publisher-3]: started with pid [17672]
[ INFO] [1552715964.748867044]: Initializing nodelet with 1 worker threads.
[ERROR] [1552715964.796146064]: Failed to load nodelet [/bebop/bebop_driver] of type [bebop_driver/BebopDriverNodelet] even after refreshing the cache: Failed to load library /home/will/bebop_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
[ERROR] [1552715964.796315733]: The error before refreshing the cache was: Failed to load library /home/will/bebop_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
[FATAL] [1552715964.796394067]: bebop_driver nodelet failed to load.
[bebop/bebop_driver-2] process has died [pid 17671, exit code 1, cmd /home/will/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/will/.ros/log/a5e7ba98-47b0-11e9-a5e9-080027ad78cc/bebop-bebop_driver-2.log].
log file: /home/will/.ros/log/a5e7ba98-47b0-11e9-a5e9-080027ad78cc/bebop-bebop_driver-2*.log

I tried changing the LD_LIBRARY_PATH but have had no success. If anyone could help, it would be much appreciated thank you!

gsilano, 2019-03-16:
Are you using a Virtual Box to run the `bebop_autonomy` package? It may be the issue.

WiPhi7600, 2019-03-17:
I am, I use a Mac and though I've seen tutorials successfully using a VM to run autonomy, I haven't seen them do it for ubuntu 18.04 and melodic

espetro, 2019-04-30:
I am getting a similar error:

> ROS_MASTER_URI=http://localhost:11311
  process[bebop/bebop_nodelet_manager-1]: started with pid [30571]
  process[bebop/bebop_nodelet-2]: started with pid [30572]
  process[bebop/robot_state_publisher-3]: started with pid [30573]
  [ INFO] [1556613283.525301409]: Initializing nodelet with 8 worker threads.
  [ERROR] [1556613283.589669967]: Failed to load nodelet [/bebop/bebop_nodelet] of type [bebop_driver/BebopDriverNodelet] even after refreshing the cache: Failed to load library /home/jterrasa/catkin_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
  [ERROR] [1556613283.589825092]: The error before refreshing the cache was: Failed to load library /home/jterrasa/catkin_ws/devel/lib//libbebop_driver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libarcommands.so: cannot open shared object file: No such file or directory)
  [FATAL] [1556613283.590231470]: Failed to load nodelet '/bebop/bebop_nodelet' of type 'bebop_driver/BebopDriverNodelet' to manager 'bebop_nodelet_manager'
  [bebop/bebop_nodelet-2] process has died [pid 30572, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load bebop_driver/BebopDriverNodelet bebop_nodelet_manager __name:=bebop_nodelet __log:=/home/jterrasa/.ros/log/a2687dd0-6b14-11e9-b2f8-a5edcd684831/bebop-bebop_nodelet-2.log].
  log file: /home/jterrasa/.ros/log/a2687dd0-6b14-11e9-b2f8-a5edcd684831/bebop-bebop_nodelet-2*.log

I am currently running `roscore` and Bebop 2 on Sphinx in the same PC as well (although the WiFi hotspot doesn't seem to get along with my hw). I am using Xubuntu 18.04 and ROS melodic.

jeremyfix, 2019-11-05:
I think this is a duplicate of #170 

For fixing your LD_LIBRARY_PATH, you have to check the path of libarcommands.so 

`$ locate libarcommands.so`

In my case, this returns 

`/opt/ros/melodic/lib/parrot_arsdk/libarcommands.so`

And then set LD_LIBRARY_PATH accordindly : 

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/ros/melodic/lib/parrot_arsdk/

temporarily for testing and then permanently in your bashrc , or zshrc , or etc..