AutonomyLab/bebop_autonomy #192: Bebop_driver error in ROS Kinetic, Ubuntu 16.04

wajeehulhassanvii, 2019-04-08:
I am trying to install the package but getting this error. It's related to bebop_driver, plz help

```
Errors     << bebop_driver:make /home/wajeeh-ros/bebop_ws/logs/bebop_driver/build.make.000.log
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function ‘bool bebop_driver::VideoDecoder::InitCodec()’:
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:5: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:5: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:26: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
                          ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:75:5: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]
     av_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:33:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavformat/avformat.h:2043:6: note: declared here
 void av_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:75:5: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]
     av_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:33:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavformat/avformat.h:2043:6: note: declared here
 void av_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:75:21: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]
     av_register_all();
                     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:33:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavformat/avformat.h:2043:6: note: declared here
 void av_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:93:36: error: ‘CODEC_CAP_TRUNCATED’ was not declared in this scope
     if (codec_ptr_->capabilities & CODEC_CAP_TRUNCATED)
                                    ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:95:32: error: ‘CODEC_FLAG_TRUNCATED’ was not declared in this scope
       codec_ctx_ptr_->flags |= CODEC_FLAG_TRUNCATED;
                                ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:97:31: error: ‘CODEC_FLAG2_CHUNKS’ was not declared in this scope
     codec_ctx_ptr_->flags2 |= CODEC_FLAG2_CHUNKS;
                               ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function ‘bool bebop_driver::VideoDecoder::ReallocateBuffers()’:
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:32: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(AV_PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5450:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
[... 83 lines cut ...]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                                                                                                    ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4771:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop_video_decoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
cd /home/wajeeh-ros/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << bebop_driver:make                    [ Exited with code 2 ]      
Failed    <<< bebop_driver                         [ 13.7 seconds ]            
Abandoned <<< bebop_tools                          [ Unrelated job failed ]    
[build] Summary: 4 of 6 packages succeeded.                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2 minutes and 38.1 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

itssme, 2019-07-15:
What version of ffmpeg do you have installed?

ibrahimqazi, 2020-02-07:
@itssme I have the same issue, mine ffmpeg version is 4.2.2, any suggestion to solve the error ?

arnaldojr, 2020-11-13:
This issue is relate to the FFMPEG headers > release 4.0.

The name of various constants got " AV_ " prefix, like AV_CODEC_CAP_TRUNCATED, AV_CODEC_FLAG_TRUNCATED, AV_CODEC_FLAG2_CHUNKS.

I submitted a pull request to #218 to fixed it.