AutonomyLab/bebop_autonomy #195: Frame size does'nt change...Cannot reduce image_width!!

ilyas95, 2019-05-31:
I want to reduce the default frame size of bebop_autonomy from 856x480 to 640x480 but it doesn't work.
I'm using bebop2 model so I went to bebop_autonomy/bebop_driver/data/bebop2_camera_calib.yaml and changed image_width from 856 to 640 but the image is still 856x480. 
In the linux console I get this error: [ERROR] [1559268611.715245116]: [Decoder] 10:10:11:715 | Decode:302 - Frame size changed to 856 x 480

I also tried to use bebop1_camera_calib.yaml that has 640 x 368 but it doesn't work neither
Someone was able to reduce the image width?

itssme, 2019-07-15:
The camera calibration file does not set the image resolution. There is a rosparam called 'PictureSettingsVideoResolutionsType'  that can be set in your bebop.launch file -> [see documentation](https://bebop-autonomy.readthedocs.io/en/latest/autogenerated/ardrone3_settings_param.html#picturesettingsvideoresolutionstype)

Here is an example of a roslaunch file that I use for my bebop. As you can see I am setting the rosparam 'PictureSettingsVideoResolutionsType' to 0 which means '1080p recording, 480p streaming'. Note that I changed the IP of my drone and I use ar_track_alvar for marker detection.

```xml
<?xml version="1.0"?>
<launch>
    <arg name="namespace" default="bebop" />
    <arg name="ip" default="192.168.1.200" />
    <arg name="drone_type" default="bebop2" /> <!-- available drone types: bebop1, bebop2 -->
    <arg name="config_file" default="$(find bebop_driver)/config/defaults.yaml" />
    <arg name="camera_info_url" default="package://bebop_driver/data/$(arg drone_type)_camera_calib.yaml" />
    <group ns="$(arg namespace)">
        <node pkg="bebop_driver" name="bebop_driver" type="bebop_driver_node" output="screen">
            <param name="camera_info_url" value="$(arg camera_info_url)" />
            <param name="bebop_ip" value="$(arg ip)" />
            <rosparam command="load" file="$(arg config_file)" />
            <rosparam>
                PictureSettingsVideoResolutionsType: 0
                PictureSettingsVideoFramerateFramerate: 2
            </rosparam>
            <param name="reset_settings" value="true" />
        </node>
        <include file="$(find bebop_description)/launch/description.launch" />
    </group>

    <arg name="marker_size" default="16.0" />
    <arg name="max_new_marker_error" default="0.04" />
    <arg name="max_track_error" default="0.4" />

    <arg name="cam_image_topic" default="/bebop/image_raw" />
    <arg name="cam_info_topic" default="/bebop/camera_info" />
    <arg name="output_frame" default="/camera_optical" />
</launch>
```

Camera calibration files are needed by nodes that use marker or tag detection and need to know which type of video stream they are processing to get a good estimate of the position. See http://wiki.ros.org/camera_calibration for more information.

gsilano, 2019-07-16:
For understanding how the camera calibration works and how to use external tools (which allow calibrating the camera using a camera, frames or video sources), I suggest taking a look at this [tutorial available on the OpenCV website](https://docs.opencv.org/4.1.0/d4/d94/tutorial_camera_calibration.html).

closed 2021-03-09 by ilyasmg