AutonomyLab/bebop_autonomy #197: Problem with Installing Bebop autonomy

LArayane, 2019-07-09:
Hello,
I got the new Bebop drone recently and I wanted to get it up and running so I followed the instructions described [Here](https://bebop-autonomy.readthedocs.io/en/latest/installation.html).
when I run: `rosdep install --from-paths src -i`
I get this:
`executing command [sudo -H apt-get install ros-kinetic-rosserial-python]
[sudo] password for rayane: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  ros-kinetic-rosserial-msgs
The following NEW packages will be installed:
  ros-kinetic-rosserial-msgs ros-kinetic-rosserial-python
0 upgraded, 2 newly installed, 0 to remove and 154 not upgraded.
Need to get 49,9 kB of archives.
After this operation, 449 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-rosserial-msgs amd64 0.7.7-0xenial-20180824-085438-0800
  404  Not Found [IP: 140.211.166.134 80]
Err:2 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-rosserial-python amd64 0.7.7-0xenial-20190320-134538-0800
  404  Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-rosserial-msgs/ros-kinetic-rosserial-msgs_0.7.7-0xenial-20180824-085438-0800_amd64.deb  404  Not Found [IP: 140.211.166.134 80]

E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-rosserial-python/ros-kinetic-rosserial-python_0.7.7-0xenial-20190320-134538-0800_amd64.deb  404  Not Found [IP: 140.211.166.134 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-rosserial-python] failed
`
Please any suggestion will be appreciated
Am using Ubuntu 16.04 and ROS Kinetic

closed 2019-07-10 by LArayane

gsilano, 2019-07-10:
Please, could you explain how did you solve the problem? Thanks!

LArayane, 2019-07-10:
The Problem was in fetching packages. I followed this  [link](https://answers.ros.org/question/325039/apt-update-fails-cannot-install-pkgs-key-not-working/) to solve it.