AutonomyLab/bebop_autonomy #200: Echoing /bebop/cmd_vel displays nothing

LArayane, 2019-08-23:
Hey,

I wanted to fly the bebop up and then make him go forward, the code is the following:

`#include<ros/ros.h>
#include<nav_msgs/Odometry.h>
#include<geometry_msgs/Twist.h>
#include<stdlib.h>
#include<math.h>

// global variable for yaw
// subscribe to navdata yaw
// in navdata yaw callback update the global variable for yaw
// use the global yaw in getpos

float tag_yaw,tag_x,tag_y,tag_z;
ros::Publisher pub;
ros::Subscriber sub;

void getpos(const nav_msgs::Odometry& posmsg){
geometry_msgs::Twist msg;
float err;

if (posmsg.pose.pose.position.z<= 1){

msg.linear.x=0.0;
msg.linear.y=0.0;
msg.linear.z=  0.05;   // exp ( 0.1);
msg.angular.x= 0;
msg.angular.y= 0;
msg.angular.z=0.0;
}

else {

msg.linear.x=0.8*(tag_x-posmsg.pose.pose.position.x);
msg.linear.y=0.0;
msg.linear.z=  0.0;   // exp ( 0.1);
msg.angular.x= 0;
msg.angular.y= 0;
msg.angular.z=0.0;
}

pub.publish(msg);
ROS_INFO_STREAM( "Showing position : "
<< "x =" << posmsg.pose.pose.position.x
<< "y =" << posmsg.pose.pose.position.y
<< "z =" << posmsg.pose.pose.position.z);
}

int main(int argc, char **argv){
ros::init(argc, argv,"velBebop");
ros::NodeHandle nh;
ros::Rate rate(2);

int i;

tag_x= 1;
tag_y= 0.0;
tag_z= 0.0;
sub = nh.subscribe("/ardrone/predictedPose",1, &getpos) ;
pub = nh.advertise<geometry_msgs::Twist>("/cmd_vel" , 1);
ros::spin();
}

`

even am not able to send commands with the command line:

`rostopic pub -r 10 /bebop/cmd_vel geometry_msgs/Twist  '{linear:  {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}`

and echoing /bebop/cmd_vel shows nothing even rqt_graph doesn't include /bebop/cmd_vel topic 

[image]
I don't know what I did wrong!! PLZ any help will be appreciated

gsilano, 2019-08-23:
I am currently using the ROS `bebop_autonomy` package to control the behavior of the Parrot Bebop 2 in Gazebo that works together with Sphinx (please check out this [link](https://developer.parrot.com/docs/sphinx/whatissphinx.html) to learn more about it). Right now, I don't know exactly what your problem is, but you could read [my code](https://github.com/gsilano/BebopS/tree/dev/sphinx) to understand how to get the drone off the ground and send the command signals.

Please note that the actual control loop is not stable. Nevertheless, you may find it useful to understand how I made the launch file and the C++ files (for controlling the drone).

LArayane, 2019-08-23:
@gsilano Thanks for your answer.
I saw your code but it is really advanced programming for me. I couldn't see where you send the commanded velocity through /bebop/cmd_vel anywhere!!

> Right now, I don't know exactly what your problem is,

my problem is that when I connect to the drone and launch the drive. I can see the topic /bebop/cmd_vel listed but when I do rqt_graph I can't find it and now I don't know how to manage controlling the drone since the topic doesn't appear!!

I want just send some velocities command to make the drone reach some point!! 
Any ideas!!!
Thanks

gsilano, 2019-08-23:
Have you taken a look at my launch files? The problem could be related to a topic remap. Look [here](https://github.com/gsilano/BebopS/blob/dev/sphinx/launch/task1_world_with_sphinx.launch) or [here](https://github.com/gsilano/BebopS/blob/dev/sphinx/launch/task2_world_with_sphinx.launch).

LArayane, 2019-08-24:
@gsilano Thanks again,
Actually it was my bad defining a wrong topic sorry. In fact, I should replace this

`sub = nh.subscribe("/ardrone/predictedPose",1, &getpos) ;
pub = nh.advertise<geometry_msgs::Twist>("/cmd_vel" , 1);`

by:

`sub = nh.subscribe("/bebop/odom",1, &getpos) ;
pub = nh.advertise<geometry_msgs::Twist>("/bebop/cmd_vel" , 1);
`
and I could publish the velocity on /bebop/cmd_vel topic like shown in this rqt_graph
[image]

but I still have a problem!! even the velocities are being published, the drone doesn't move at all !!!
Please any suggestions what could be the problem, Thanks.

LArayane, 2019-08-27:
I solved the problem. by just send an empty message to /bebop/takeoff topic before running the node.
Thanks for your help 
I have another question!!! where to get the battery state so I can send an error message when the battery is low!!!
Please help

gsilano, 2019-08-27:
I didn't get your point. What do you want exactly? 1) Receive an error message when the battery level is too low 2) Use the battery status information to design a controller that manages the landing?
As I know, the Parrot Bebop 2 onboard firmware already manages it. When the battery level is too low (I don't remember exactly the battery percentage) the drone stops the task executions and lands on the ground.

LArayane, 2019-08-27:
@gsilano I have parrot bebop. and I want to use the battery status information to design a controller that manages the landing!!
How can I do that!! and on which topic this information is available!!
thanks in advance

gsilano, 2019-08-27:
[Here](https://bebop-autonomy.readthedocs.io/en/latest/index.html) you find all the information that you need. Furthermore, do not forget to use the issue tracker to read questions and answers on topics already discussed.

LArayane, 2019-08-29:
@gsilano how can I use issue tracker!! and what's to track!! Sorry I search and I found nothing

gsilano, 2019-08-30:
I suggest you to start reading [this page](https://help.github.com/en/articles/about-issues), and then the others linked reported into the conversation. Moreover, I think you could use these ROS services to track the drone battery status and to get the information you need.

[CommonChargerStateCurrentChargeStateChanged](https://bebop-autonomy.readthedocs.io/en/latest/autogenerated/common_states_param_topic.html?highlight=battery#commonchargerstatecurrentchargestatechanged)
[Ardrone3PilotingStateAlertStateChanged](https://bebop-autonomy.readthedocs.io/en/latest/autogenerated/ardrone3_states_param_topic.html?highlight=battery#ardrone3pilotingstatealertstatechanged)
[Ardrone3MediaRecordStateVideoStateChangedV2](https://bebop-autonomy.readthedocs.io/en/latest/autogenerated/ardrone3_states_param_topic.html?highlight=battery#ardrone3mediarecordstatevideostatechangedv2)
[CommonCommonStateBatteryStateChanged](https://bebop-autonomy.readthedocs.io/en/latest/autogenerated/common_states_param_topic.html?highlight=battery#commoncommonstatebatterystatechanged)

LArayane, 2019-08-30:
@gsilano Thanks. I will do it!
Another question please, I want to get the simulation time using /clock [topic](http://wiki.ros.org/Clock) 
but this topic is missing when I list the topics even the parameter use_sim_time doesn't exist!!
My purpose is to generate a sinosoidal trajectory with cos( pi t) and the time t here has to be set to 0 when the simulation starts.

Any suggestion please !!

gsilano, 2019-08-30:
I think, but I could be wrong, you have to develop your trajectory generator using an external C++ file. In this file, your trajectory reference, in terms of x, y, and z components, is sent to the Barropt Bebop in order to generate the appropriate command signals. This is an idea, and I don't if it works or not, so you have to try.

LArayane, 2019-08-30:
@gsilano ok thanks I will try :)

LArayane, 2019-09-01:
> I solved the problem. by just send an empty message to /bebop/takeoff topic before running the node.
> Thanks for your help
> I have another question!!! where to get the battery state so I can send an error message when the battery is low!!!
> Please help

closed 2019-09-01 by LArayane