AutonomyLab/bebop_autonomy #203: Problem with calibrating the camera!!

LArayane, 2019-09-04:
Hi,
I need the projection matrix K to convert pixels to meters. I have calibrated the camera and actually I got the same values as in the default file but even so I am still getting the same error

    [ERROR] [1567353696.489721846]: [Decoder] 18:01:36:489 | Decode:302 - Frame size changed to 856 x 480

and even if I change the depth and height and set them to a different values than 856 x 480 I get the same error with same values!!!
How can I solve this please I think the matrix K is critical in my work. Thanks

This is what I get after calibration:

`('D = ', [0.01855441773892149, -0.04992192012885487, -0.007079594166185704, 0.002927466912916301, 0.0])
('K = ', [534.9749658295674, 0.0, 432.80365705109926, 0.0, 523.7279535777849, 227.29982426950724, 0.0, 0.0, 1.0])
('R = ', [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0])
('P = ', [530.457763671875, 0.0, 436.44346924338606, 0.0, 0.0, 524.9320068359375, 224.57206054203925, 0.0, 0.0, 0.0, 1.0, 0.0])
None
# oST version 5.0 parameters

[image]

width
856

height
480

[narrow_stereo]

camera matrix
534.974966 0.000000 432.803657
0.000000 523.727954 227.299824
0.000000 0.000000 1.000000

distortion
0.018554 -0.049922 -0.007080 0.002927 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
530.457764 0.000000 436.443469 0.000000
0.000000 524.932007 224.572061 0.000000
0.000000 0.000000 1.000000 0.000000

('Wrote calibration data to', '/tmp/calibrationdata.tar.gz')
('D = ', [0.01855441773892149, -0.04992192012885487, -0.007079594166185704, 0.002927466912916301, 0.0])
('K = ', [534.9749658295674, 0.0, 432.80365705109926, 0.0, 523.7279535777849, 227.29982426950724, 0.0, 0.0, 1.0])
('R = ', [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0])
('P = ', [530.457763671875, 0.0, 436.44346924338606, 0.0, 0.0, 524.9320068359375, 224.57206054203925, 0.0, 0.0, 0.0, 1.0, 0.0])
# oST version 5.0 parameters

[image]

width
856

height
480

[narrow_stereo]

camera matrix
534.974966 0.000000 432.803657
0.000000 523.727954 227.299824
0.000000 0.000000 1.000000

distortion
0.018554 -0.049922 -0.007080 0.002927 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
530.457764 0.000000 436.443469 0.000000
0.000000 524.932007 224.572061 0.000000
0.000000 0.000000 1.000000 0.000000`

gsilano, 2019-09-04:
Hi @LArayane! To calibrate the camera, you can use the OpenCV features, as is usually done with RGB cameras. [Here,](https://docs.opencv.org/3.4.7/dc/dbb/tutorial_py_calibration.html) you find a useful guide.

LArayane, 2019-09-04:
Thanks @gsilano  for the answer!
Actually I used this [link](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) to do calibration and I replaced the .yaml files in /bebop_driver/data/bebop1_camera_calib.yaml and /bebop_driver/data/bebop2_camera_calib.yaml although I wonder why there is two yaml files!!!
and even by doing that I still get the error!! and even if I change the values in the .yaml files the error stays the same with the same values!! So I don't think the calibration parameters are taken from the .yaml files!!

Any suggestion please or clarification!! thanks

gsilano, 2019-09-04:
I do not know how the suggested application in the link works, therefore, based on my experience, if I had to calibrate the Bebop 2 camera I would prefer to use the well known OpenCV function. As you can read, the code can also be used offline. You can record a video (it is mandatory to have a chessboard) and then process it using the code provided in the link I gave you. 

You should investigate doing reverse engineering on the `bebop_autonomy` ROS package.

LArayane, 2019-09-04:
For the package I used, in [here](http://wiki.ros.org/camera_calibration) they said that it uses OpenCV camera calibration, described [here](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html). but no worries I can do both and I will let you know about the result if something changes.
Thanks