AutonomyLab/bebop_autonomy #205: Problem while launching bebop_driver

SekenWang, 2019-11-11:
I have search the issue and there seems nobody had faced this problem.I try to run sphinx and bebop_autonomy on the same PC, but I always get the same error that it seems cant connect between the process.
Begging for your help,and best regards.
Btw, here is my launch file , drone file together with the terminal report:

**launch:**

`<?xml version="1.0"?>
<launch>
    <arg name="namespace" default="bebop_leader" />
    <arg name="ip" default="10.202.0.1" />
    <arg name="drone_type" default="bebop2" /> 
    <arg name="config_file" default="$(find bebop_driver)/config/defaults.yaml" />
    <arg name="camera_info_url" default="package://bebop_driver/data/$(arg drone_type)_camera_calib.yaml" />
    <group ns="$(arg namespace)">
        <node pkg="bebop_driver" name="bebop_driver" type="bebop_driver_node" output="screen">
            <param name="camera_info_url" value="$(arg camera_info_url)" />
            <param name="bebop_ip" value="$(arg ip)" />
            <rosparam command="load" file="$(arg config_file)" />
        </node>
        <include file="$(find bebop_description)/launch/description.launch" />
    </group>
</launch>`

**drone**:

<?xml version="1.0" encoding="UTF-8"?>
<drone
  name="bebop_leader"
  firmware="/home/seken/mycode/workspace/drl-landing/src/bebop_autonomy/bebop_description/model/ardrone3-milos_pc.ext2.zip"
  hardware="milosboard">
  <machine_params
    low_gpu="ture"
    with_front_cam="true"
    with_hd_battery="false"
    with_flir="false"
    flir_pos="tilted"/>
  <pose>default</pose>
  <interface>eth1</interface>
  
  <stolen_interface>wlan0:eth0:192.168.42.1/24</stolen_interface>
</drone>

and here is the terminal report:
**SUMMARY**

PARAMETERS
 * /bebop_leader/bebop_driver/bebop_ip: 10.202.0.1
 * /bebop_leader/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop_leader/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop_leader/bebop_driver/odom_frame_id: odom
 * /bebop_leader/bebop_driver/publish_odom_tf: True
 * /bebop_leader/bebop_driver/reset_settings: True
 * /bebop_leader/bebop_driver/states/enable_altitudechanged: True
 * /bebop_leader/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop_leader/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop_leader/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop_leader/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop_leader/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop_leader/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop_leader/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop_leader/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop_leader/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop_leader/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop_leader/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop_leader/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop_leader/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop_leader/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop_leader/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop_leader/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop_leader/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
[... 41 lines cut ...]
[ INFO] [1573469382.671548771]: [CB] 18:49:42:671 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1118 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1573469382.672624363]: [CB] 18:49:42:672 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1183 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1573469382.692279805]: [CB] 18:49:42:692 | CommonFlightPlanStateAvailabilityStateChanged:1568 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1573469382.694484296]: [CB] 18:49:42:694 | CommonFlightPlanStateComponentStateListChanged:1619 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1573469382.718019180]: [ARCONTROLLER_Network] 18:49:42:717 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1573469382.718183358]: [ARCONTROLLER_Stream2] 18:49:42:718 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1573469382.718211631]: [ARCONTROLLER_Stream2] 18:49:42:718 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ERROR] [1573469382.718337611]: [ARDISCOVERY_Connection] 18:49:42:718 | ARDISCOVERY_Connection_ControllerInitSocket:669 - connect() SO_ERROR: 111 Connection refused
[ERROR] [1573469382.718385632]: [ARDISCOVERY_Device] 18:49:42:718 | ARDISCOVERY_Device_NewARNetworkAL:243 - error: Socket access permission denied
[ERROR] [1573469382.718405727]: [ARCONTROLLER_Network] 18:49:42:718 | ARCONTROLLER_Network_New:355 - error: Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1573469382.718486245]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StartRun:5949 - Start failed or canceled.
[ERROR] [1573469382.718526308]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StopNetwork:7264 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_GENER
[ERROR] [1573469382.718542128]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StopNetwork:7282 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_ARDRO
[ERROR] [1573469382.718557382]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StopNetwork:7291 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_COMMO
[ERROR] [1573469382.718570318]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StopRun:5992 - StopNetwork failed with error :Bad parameters
[ INFO] [1573469382.718695764]: [BebopSDK] 18:49:42:718 | Cleanup:326 - Bebop Cleanup()
[ERROR] [1573469382.718731714]: [ARCONTROLLER_Device] 18:49:42:718 | ARCONTROLLER_Device_StartRun:5957 - Start fail error :Error during the getting of the ARNETWORKAL_Manager from the device
[FATAL] [1573469382.718938270]: Init failed: Waiting for device failed: No error
[ INFO] [1573469382.719029007]: Bebop Nodelet Dstr: 0
[ INFO] [1573469382.719044145]: Killing Camera Thread ...
[ INFO] [1573469382.719058504]: Killing Aux Thread ...
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[bebop_leader/bebop_driver-2] process has died [pid 22820, exit code -6, cmd /home/seken/mycode/workspace/drl-landing/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/seken/.ros/log/f6cdf364-0470-11ea-8b5b-40b0344f6594/bebop_leader-bebop_driver-2.log].
log file: /home/seken/.ros/log/f6cdf364-0470-11ea-8b5b-40b0344f6594/bebop_leader-bebop_driver-2*.log

SekenWang, 2019-11-25:
begging for help, please!

fbrunodr, 2020-06-30:
Did you solve it? I have the same problem.

EricDiazR, 2021-04-12:
Did you manage to make it work? I have the same problem :(

diligentYing, 2024-04-10:
I meet the same problem. I find its reason is IP_address. You can set bebop's IP as the defult IP: "192.168.42.1". By doing so, I resovled this problem.

diligentYing, 2024-04-10:
I meet the same problem. I find its reason is IP_address. You can set bebop's IP as the defult IP: "192.168.42.1" in your launch file. By doing so, I resovled this problem.