AutonomyLab/bebop_autonomy #206: How to save Images from bebop forward camera!!

LArayane, 2020-01-13:
Hi,

Am working on depth observation and I need to save images from bebop forward camera. These images are to be processed offline!! Can anybody help me! Thanks

Rayane

cesarhcq, 2020-01-15:
If you're using ROS, you need to see what's topic from Bebop images, for example:
/bebop/image_raw

You can execute the following command to save all topics from Bebop: $ rosbag record -a

However, if you're intend to save a specific topic, you can run: $ rosbag record /bebop/image_raw/

See more in: [Rosbag CommandLine](http://wiki.ros.org/rosbag/Commandline) and [Rosbag Export Images](http://wiki.ros.org/rosbag/Tutorials/Exporting%20image%20and%20video%20data)

LArayane, 2020-02-17:
Thanks for the answer @cesarhcq  but it takes a lot of time converting bag file to txt file. maybe for other topics. am intending to save them not for a replay. is there any other way out