AutonomyLab/bebop_autonomy #215: roslaunch bebop_node.launch fails. [FATAL] [1596508408.108566618]: Init failed: Waiting for device failed: No error

zpp1944540154, 2020-08-04:
roslaunch bebop_node.launch fails.
The error is shown:
... logging to /home/zpp/.ros/log/dec24eee-d5fa-11ea-b958-f8cab82c6e66/roslaunch-zpp123-11985.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zpp123:34811/

SUMMARY
PARAMETERS
 * /bebop/bebop_driver/PilotingSettingsMaxTiltCurrent: 15
 * /bebop/bebop_driver/SpeedSettingsMaxRotationSpeedCurrent: 180
 * /bebop/bebop_driver/SpeedSettingsMaxVerticalSpeedCurrent: 1
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [11998]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dec24eee-d5fa-11ea-b958-f8cab82c6e66
process[rosout-1]: started with pid [12010]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [12017]
process[bebop/robot_state_publisher-3]: started with pid [12018]
[ INFO] [1596508406.701823128]: Initializing nodelet with 4 worker threads.
[ INFO] [1596508408.054998982]: [BebopSDK] 10:33:28:054 | Bebop:225 - Bebop Cnstr()
[ INFO] [1596508408.055110741]: Nodelet Cstr
[ INFO] [1596508408.060246595]: Connecting to Bebop ...
[ INFO] [1596508408.062789985]: [CB] 10:33:28:062 | Ardrone3PilotingStateFlatTrimChanged:388 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1596508408.063928090]: [CB] 10:33:28:063 | Ardrone3PilotingStateFlyingStateChanged:432 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1596508408.065304262]: [CB] 10:33:28:065 | Ardrone3PilotingStateNavigateHomeStateChanged:534 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1596508408.066426694]: [CB] 10:33:28:066 | Ardrone3PilotingStatePositionChanged:592 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1596508408.068003506]: [CB] 10:33:28:067 | Ardrone3PilotingStateSpeedChanged:657 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1596508408.069728600]: [CB] 10:33:28:069 | Ardrone3PilotingStateAttitudeChanged:722 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1596508408.071246666]: [CB] 10:33:28:071 | Ardrone3PilotingStateAltitudeChanged:838 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1596508408.075138084]: [CB] 10:33:28:075 | Ardrone3MediaStreamingStateVideoEnableChanged:1388 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1596508408.076544883]: [CB] 10:33:28:076 | Ardrone3CameraStateOrientation:1490 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1596508408.079844954]: [CB] 10:33:28:079 | Ardrone3GPSStateNumberOfSatelliteChanged:1882 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1596508408.085337728]: [CB] 10:33:28:085 | CommonCommonStateBatteryStateChanged:148 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1596508408.088715208]: [CB] 10:33:28:088 | CommonCommonStateWifiSignalChanged:510 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1596508408.092462341]: [CB] 10:33:28:092 | CommonOverHeatStateOverHeatChanged:1023 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1596508408.093863396]: [CB] 10:33:28:093 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1118 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1596508408.094875274]: [CB] 10:33:28:094 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1183 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1596508408.098036148]: [CB] 10:33:28:097 | CommonFlightPlanStateAvailabilityStateChanged:1568 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1596508408.099286073]: [CB] 10:33:28:099 | CommonFlightPlanStateComponentStateListChanged:1619 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1596508408.106984401]: [ARCONTROLLER_Network] 10:33:28:106 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1596508408.107252000]: [ARCONTROLLER_Stream2] 10:33:28:107 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1596508408.107297204]: [ARCONTROLLER_Stream2] 10:33:28:107 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ERROR] [1596508408.107857133]: [ARDISCOVERY_Connection] 10:33:28:107 | ARDISCOVERY_Connection_ControllerInitSocket:669 - connect() SO_ERROR: 111 Connection refused
[ERROR] [1596508408.107929398]: [ARDISCOVERY_Device] 10:33:28:107 | ARDISCOVERY_Device_NewARNetworkAL:243 - error: Socket access permission denied
[ERROR] [1596508408.107962760]: [ARCONTROLLER_Network] 10:33:28:107 | ARCONTROLLER_Network_New:355 - error: Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1596508408.108059460]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StartRun:5949 - Start failed or canceled.
[ERROR] [1596508408.108114631]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StopNetwork:7264 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_GENER
[ERROR] [1596508408.108145047]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StopNetwork:7282 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_ARDRO
[ERROR] [1596508408.108180439]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StopNetwork:7291 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_COMMO
[ERROR] [1596508408.108205293]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StopRun:5992 - StopNetwork failed with error :Bad parameters
[ERROR] [1596508408.108237550]: [ARCONTROLLER_Device] 10:33:28:108 | ARCONTROLLER_Device_StartRun:5957 - Start fail error :Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1596508408.108367054]: [BebopSDK] 10:33:28:108 | Cleanup:326 - Bebop Cleanup()
[FATAL] [1596508408.108566618]: Init failed: Waiting for device failed: No error
[ INFO] [1596508408.108664117]: Bebop Nodelet Dstr: 0
[ INFO] [1596508408.108689558]: Killing Camera Thread ...
[ INFO] [1596508408.108722174]: Killing Aux Thread ...
[bebop/bebop_driver-2] process has died [pid 12017, exit code -11, cmd /home/zpp/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/zpp/.ros/log/dec24eee-d5fa-11ea-b958-f8cab82c6e66/bebop-bebop_driver-2.log].
log file: /home/zpp/.ros/log/dec24eee-d5fa-11ea-b958-f8cab82c6e66/bebop-bebop_driver-2*.log

 Could someone please help me for solving this bug? Thanks.

zpp1944540154, 2020-08-04:
I meet this bug twice using Ubuntu16 and Ubuntu18. And I cannot solve it by myself. Please give me some suggests.

Canonvn, 2020-08-06:
I'm having trouble with the same error.

hychong1997, 2020-09-15:
I have the same problem. Does anyone konw the reason?