AutonomyLab/bebop_autonomy #222: Can this be installed on ros noetic?

aliwaqas333, 2021-07-10:
I am facing some issues when installing. especially when i run the rosdep command, 
`bebop_driver: Cannot locate rosdep definition for [parrot_arsdk]` , i dont know what that means, can someone kindly help me. 

I am using ros noetic with ubuntu 20.04.

mbakr99, 2021-09-01:
I am facing the same issue. However, I think you should be using the ros Idigo, Jade, or Kinetic as per the documentation. I pulled a ros indigo image on a docker container to avoid changing the OS version.

EDU4RDO-SH, 2021-09-06:
@aliwaqas333 Were you able to solve the problem?

aliwaqas333, 2021-11-02:
This can't be installed on ubuntu 20.04.

arnaldojr, 2021-11-02:
I'm using the Hector quadrotor in my simulations like a alternative in ROS Noetic.  

[hector quadrotor noetic](https://github.com/arnaldojr/hector_quadrotor_noetic  )

oli06, 2022-01-20:
I was able to install Bebop_autonomy using ROS Noetic inside Ubuntu 20.04 using this [ParrotArsdk wrapper](https://github.com/antonellabarisic/parrot_arsdk/tree/noetic_dev). See their README.md for installation instructions.

AldrichCabrera, 2022-02-08:
Thank you @oli06 It works for me.