AutonomyLab/bebop_autonomy #224: Can't check drone velocities.

guiaugustoga987, 2023-08-04:
Hi,

I need to know the drone velocities in m/s. How do I acess this information ? If I echo `/bebop/odom` topic I get all twist.twist data = 0 as you can see :

header: 
  seq: 522
  stamp: 
    secs: 1691184697
    nsecs: 428244013
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: 0.0
      y: 0.0
      z: 0.0
    orientation: 
      x: -0.0007713166171202364
      y: -0.017880132644812745
      z: -0.9997898473865937
      w: 0.009998349362347272
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: -0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0

On topic `/bebop/states/ardrone3/PilotingState/SpeedChanged` I get : 

---
header: 
  seq: 868
  stamp: 
    secs: 1691183742
    nsecs: 303088109
  frame_id: "base_link"
speedX: 0.0
speedY: 0.0
speedZ: 0.0
---

How do I access these data ?

Thanks.

guiaugustoga987, 2023-08-04:
It seems like the values are only published when a takeoff command is send to the drone.

araujorayza, 2023-08-17:
I've had a similar problem but no response in #187 . Are you using bebop 2?

guiaugustoga987, 2023-08-17:
Yes. As I said, the linear velocities display non-zero values only when the drone is flying. But the angular velocities remain zero anyway.